tum-vision/lsd_slam #342: Made docker image of LSD-SLAM for anyone who are struggling with build error.

tigerk0430, 2019-12-25:
run as follows. (tested on Ubuntu 16.04 LTS)

$ `docker pull edward0im/starlaw:lsd-slam`
$ `vim run_lsd_slam.sh`
type below
```
#!/bin/sh
XSOCK=/tmp/.X11-unix
XAUTH=/home/${USER}/.Xauthority

xhost +local:docker

	# --runtime=nvidia \
docker run \
	--net=host \
	--name lsdslam \
	--rm \
    -it \
	--env="XAUTHORITY=${XAUTH}" \
	--env="DISPLAY=unix${DISPLAY}" \
	--env="XDG_RUNTIME_DIR=/run/user/1000" \
	--privileged  \
	-v /run/user/1000:/run/user/1000 \
	-v /dev/bus/usb:/dev/bus/usb \
	-v ${XSOCK}:${XSOCK}:rw \
	-v ${XAUTH}:${XAUTH}:rw \
	edward0im/starlaw:lsd-slam \
	/root/run_lsdslam.sh
```
$ `chmod 777 run_lsd_slam.sh`
$ `./run_lsd_slam.sh`

if anyone who has got the error while following those steps, comment  below.

ibrahimLearning, 2020-06-22:
would you provide your docker file?

edward0im, 2020-06-22:
@ibrahimLearning 

Unfortunately, I don't have a Dockerfile of this image. I had built lsd-slam manually while using docker image. 
You can find the image in https://hub.docker.com/repository/docker/edward0im/starlaw

damonftl, 2020-06-24:
I'm getting this error:

QGLTempContext: No GL capable X visuals available.

Running "glxinfo" doesn't give any errors, and glxgears runs just fine. Full disclosure though: I'm running this in a virtual machine with VirtualBox running Ubuntu 20.04, so there is no direct access to the GPU. Not sure how to fix this problem or if it is even possible to fix it with my setup.

edward0im, 2020-06-25:
hello @damonftl,

could you check Ubuntu version for this docker image? because It was made in Ubuntu 16.04. 
If you have a trouble while running on Ubuntu 16.04, I can help you :-)

kareotoko, 2020-09-28:
Hi @edward0im 
I am following your step above. But I encountered a problem:
I implement it on Native Ubuntu 16.04 LTS
$`./run_lsd_slam.sh`
`non-network local connections being added to access control list
/root/run_lsdslam.sh: line 4: kill: %1: no such job
Started QApplication thread
/root/run_lsdslam.sh: line 6:    76 Illegal instruction     (core dumped) rosrun lsd_slam_core live_slam image:=/camera/image_rect camera_info:=/camera/camera_info`
Do you know where the problem is?

edward0im, 2020-09-28:
Hi @kareotoko,

I've upgraded Ubuntu 16.04 to 18.04 six months ago and I didn't take care of this docker image.
And recently, docker commands have been changed a lot because of the docker version upgrade, but I couldn't follow up on the newest command.

I'm sorry but I'll upload LSD-SLAM docker image running on Ubuntu 18.04 LTS within a week. 
Therefore, LSD-SLAM docker image running on Ubuntu 16.04 LTS will not be on update. Sorry about that.