tum-vision/lsd_slam #343: Compilation fails under Melodic

chrisspen, 2020-03-15:
I'm trying to install it under ROS Melodic and Ubuntu 18.04 but the `rosmake lsd_slam` command is failing with the error:

      Traceback (most recent call last):
        File "/opt/ros/melodic/share/dynamic_reconfigure/cmake/gendeps", line 66, in <module>
          imp.load_module("__main__", f, srcfile, ('.cfg', 'U', 1))
        File "~/git/slam_test/test_lsdslam/catkin_ws/src/lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg", line 20, in <module>
          gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on point's variance, in the respective keyframe's scale. ", -3, -10, 1)
        File "/opt/ros/melodic/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 274, in add
          self.group.add(name, paramtype, level, description, default, min, max, edit_method)
        File "/opt/ros/melodic/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 145, in add
          check_description(description)
        File "/opt/ros/melodic/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 69, in check_description
          raise Exception(r"""quotes not allowed in description string `%s`""" % description)
      Exception: quotes not allowed in description string `log10 of threshold on point's variance, in the respective keyframe's scale. `
      
      CMake Error at /opt/ros/melodic/share/dynamic_reconfigure/cmake/cfgbuild.cmake:78 (string):
        string sub-command REPLACE requires at least four arguments.
      Call Stack (most recent call first):
        /opt/ros/melodic/share/dynamic_reconfigure/cmake/cfgbuild.cmake:99 (gencfg_cpp)
        CMakeLists.txt:29 (include)
      
      
      -- [rosbuild] Found SSE3 extensions, using flags: -msse3 -mfpmath=sse
      -- Configuring incomplete, errors occurred!
      See also "~/git/slam_test/test_lsdslam/catkin_ws/src/lsd_slam/lsd_slam_viewer/build/CMakeFiles/CMakeOutput.log".
      See also "~/git/slam_test/test_lsdslam/catkin_ws/src/lsd_slam/lsd_slam_viewer/build/CMakeFiles/CMakeError.log".
      /opt/ros/melodic/share/mk/cmake.mk:7: recipe for target 'all' failed

Is the package supported on Ubuntu 18.04 and Melodic? If so, how do I fix this error?

thuliangjz, 2020-04-16:
as it said just delete those single quote, i.e. ' in 
1) lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg
2) lsd_slam_core/cfg/LSDDebugParams.cfg

some other issues you might come across are posted on this [blog](https://blog.csdn.net/softimite_zifeng/article/details/78632653)though I still have problems unsolved yet

mirellameelo, 2020-07-20:
@chrisspen @thuliangjz Did you succeed compilation on Melodic?

thuliangjz, 2020-07-24:
@mirellameelo I managed to compiled the project with this fork https://github.com/kevin-george/lsd_slam, the oringinal one seems to be not maintained for some time.